raspberrypi/raspiraw #2: "-b" option does not work for 8 (OK for 10, 12 and 16)

Hermann-SW, 2017-12-31:
$ gdb raspiraw
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
...
Reading symbols from raspiraw...done.
(gdb) b 893
Breakpoint 1 at 0x12ecc: file raspiraw.c, line 893.
(gdb)

(gdb) r -md 7 -b 8
Starting program: /home/pi/raspiraw.6by9/raspiraw -md 7 -b 8
[Thread debugging using libthread_db enabled]
...
mmal: mmal_component_enable: vc.ril.video_render 2

Thread 1 "raspiraw" hit Breakpoint 1, main (argc=5, argv=0xbefff6c4) at raspiraw.c:893
893		status = mmal_port_format_commit(output);
(gdb)

"buffer_size_recommended" used later gets set to a crazy 459MB value:
Code: Select all

(gdb) p output->buffer_size_recommended
$1 = 384000
(gdb) n
mmal: mmal_port_format_commit: vc.ril.rawcam(3:0) port 0x28fb0 format 3:GBRG
mmal: mmal_vc_port_info_set: set port info (3:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
894		if(status != MMAL_SUCCESS)
(gdb) p output->buffer_size_recommended
$2 = 481996800
(gdb) 

(gdb) p /x output->buffer_size_recommended
$4 = 0x1cbab000
(gdb)

The size I would expect for 640x480 -b8 is 0x4B000:
$ echo "obase=16; 640*480" | bc -q
4B000
$ 

" buffer_size_min", "buffer_size_recommended" as well as "buffer_size" are affected:
(gdb) n
mmal: mmal_port_format_commit: vc.ril.rawcam(3:0) port 0x28fb0 format 3:GBRG
mmal: mmal_vc_port_info_set: set port info (3:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
894		if(status != MMAL_SUCCESS)
(gdb) p *output
$7 = {priv = 0x28ff0, name = 0x29278 "vc.ril.rawcam:out:0", type = MMAL_PORT_TYPE_OUTPUT, index = 0, index_all = 1, 
  is_enabled = 0, format = 0x289b0, buffer_num_min = 4, buffer_size_min = 481996800, buffer_alignment_min = 0, 
  buffer_num_recommended = 4, buffer_size_recommended = 481996800, buffer_num = 4, buffer_size = 481996800, 
  component = 0x28720, userdata = 0x0, capabilities = 2}
(gdb)

6by9, 2018-01-02:
What version of firmware are you running?
```vcgencmd version```

6by9, 2018-01-02:
Taking the latest rpi-update firmware (Dec  1 2017 16:18:56 38ac72569d7fbd7cf75754d9f14c2495416aaee7) and running ```./raspiraw -b 8 -md 0 -y 4 -o /dev/null``` works fine for me (-y 4 as I have some strange I2C bus setups).

Using the latest version from Raspbian (Oct 24 2017 17:12:00 a3d7660e6749e75e2c4ce4d377846abd3b3be283) also works quite happily, allocating a 8121344 byte buffer for the 8MPix image for IMX219 mode 0.

There was a fix back in July that addressed an issue with stride calculations. I'm really hoping you're on something more recent than that. https://github.com/Hexxeh/rpi-firmware/commit/a4e1434bfa3fe9daf7413e0d78b83afa236e102b

Hermann-SW, 2018-01-05:
> thread [2].
> Links:

pi@raspberrypi01:~/raspiraw/t $ vcgencmd version
Oct 24 2017 17:13:32
Copyright (c) 2012 Broadcom
version a3d7660e6749e75e2c4ce4d377846abd3b3be283 (clean) (release)
pi@raspberrypi01:~/raspiraw/t $

Hermann.

Hermann-SW, 2018-01-05:
> that. Hexxeh/rpi-firmware@a4e1434 [1]
> thread [3].

Hi,

as sent with previous email I use the October version you use.
But I raised the issue with "-b 8" for "-md 7", you tried with "-md 0".
Please try with "-md 7" and you will see the issue.
Or look into the issue gdb session I provided.

Hermann.

Hermann-SW, 2018-01-06:
you need a v1 camera for "-md 7" since modes 1-7 are available to v1 cameras currently only with raspiraw

6by9, 2018-01-09:
There's no point in restricting yourself to one specific version of the firmware - things move on. 

I'd actually fixed this problem back in November, but it only got released yesterday in the firmware update.
https://github.com/raspberrypi/userland/blob/master/interface/mmal/util/mmal_il.c#L727 only lists SBGGR8, not the other Bayer orders. The firmware uses that function (built for the VPU) to convert between MMAL and IL encodings, so RGGB, GBRG, and GRBG will fail.
Syncing back to the official Raspbian firmware (Oct 24th), using -hf to select BGGR works. On the latest rpi-update code all Bayer orders work.

Hermann-SW, 2018-01-11:
Thank you so much!
I had to learn that "sudo apt-get update" and "sudo apt-get upgrade" was not enough.
But after "sudo rpi-update" and reboot I am current:
`pi@raspberrypi2B:~ $ vcgencmd version
Jan  8 2018 14:33:26 
Copyright (c) 2012 Broadcom
version cc4313d3130326f2ce45d951a3c78b7a8bf20ad3 (clean) (release)
pi@raspberrypi2B:~ $ `

Now this command works flawless:
`raspiraw -md 7 -b 8 -o out.%04d.raw`

All out.*.raw files now have size 307200 (which is 640*480).
And those out.*.raw files can be converted to gray8 .png easily:
`$ cat <(echo -e 'P5\n640 480\n255') out.0401.raw | pgmtoppm | pnmtopng >out.0401.raw.pgm.png
$ `

Nice!
[image]

closed 2018-01-11 by Hermann-SW