raspberrypi/raspiraw #5: [Help needed] RaspiRaw: No sensor found. Aborting

CarlosGS, 2018-02-24:
Hi! First of all: thank you so much @6by9 and @Hermann-SW for sharing all your hard work that allows to use these camera sensors to their full capacity :-)

I'm trying to get the provided examples to work in my setup, and getting the following output:
```
pi@raspberrypi:~ $ camera_i2c 
setting GPIO for board revsion: a02082
Raspberry Pi3
Set state of 133 to 1
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
pi@raspberrypi:~ $ raspiraw -md 7 -t 1000 -ts tstamps.csv -hd0 hd0.32k -h 64 --vinc 1F --fps 660 -r "380A,0040;3802,78;3806,0603" -sr 1 -o /dev/shm/out.%04d.raw
Using i2C device /dev/i2c-0
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: No sensor found. Aborting
pi@raspberrypi:~ $ 
```

So the question I'm here to ask: **I thought I had a ov5647 sensor, but how can I be sure?**
[image]

Thanks again, and cheers!

Terminus-IMRC, 2018-02-24:
Try putting `dtparam=i2c_vc=on` line to your `config.txt` file.

CarlosGS, 2018-02-24:
Ohh that was it!
The config.txt said `dtparam=i2c_arm=on` so just needed to replace "arm" with "vc".

I was also stuck with "Cannot decode file out.X.raw", this was caused by using the "dcraw" program in raspbian repositories, instead of https://github.com/6by9/dcraw (which solved it).

Here is a video of the result:
[image]

Thanks for the super-fast reply!! :-)

closed 2018-02-24 by CarlosGS

JSEsq, 2018-08-28:
Hi,
i have a camera v.2.1 (see picture), and i have the same problem (**no sensor found**).

I tried both variants, dtparam=i2c_arm=on and dtparam=i2c_vc=on, with the same result.

below are some commands and outputs:

pi@raspi:~/raspiraw $ ./raspiraw -md 7 -t 1000 --height 480 --top 0 -sr 1 -g 240 -e 200 --vinc 17 -o /dev/shm/o ut.%04d.raw
Using i2C device /dev/i2c-0
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: No sensor found. Aborting

pi@raspi:~/raspiraw $ vcgencmd get_camera
supported=1 detected=1

pi@raspi:~/raspiraw $ raspistill -t 1000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

raspiraw v.0.0.2

**Is there any ideas how to fix it?**

[image]

6by9, 2018-08-29:
0x64 is the crypto chip.
0x10 would be the sensor.
Have you run ./camera_i2c to set up the shutdown GPIOs to the camera correctly, as per the instructions? https://github.com/6by9/raspiraw#raspiraw-usage

And posting on a closed issue is not a good way to request help - most people won't look at it.

kamomil, 2020-04-24:
> 0x64 is the crypto chip.
Hi, what exactly is the crypto chip?
When plugging the camera away the 0x64 is not detected anymore from the i2cdetect

> 0x10 would be the sensor.
> Have you run ./camera_i2c to set up the shutdown GPIOs to the camera correctly, as per the instructions? https://github.com/6by9/raspiraw#raspiraw-usage
> 
> And posting on a closed issue is not a good way to request help - most people won't look at it.

6by9, 2020-04-24:
The IMX219 module sold by Raspberry Pi has a crypto chip onboard to avoid cloning. This is checked by the firrmware before it will allow the sensor to be used with the firmware camera stack.

With OV5647 a number of third parties made clones of the sensor board with alternate lenses, or just cloning and claiming to be Raspberry Pi cameras. This lead rise to a fair number of complaints over image quality which were not down to RPT.
Those with alternate lenses often hadn't considered lens shading compensation, so users complained of vignetting.

With IMX219 the crypto chip was added to avoid this situation.

raspiraw bypasses this anyway as the firmware is not driving the sensor. Any image quality issues are most likely down to dodgy control algorithms for AE or AWB.