raspberrypi/raspiraw #10: Cannot decode captured video file

kunguz, 2018-05-15:
Hi,

I am having this trouble in decoding the capture data to a video file on my raspberry pi zero w. 

This is what I got in the bash:
``
pi@muska:~/raspiraw/tools $ ./320x240 1000
removing /dev/shm/out.*.raw
capturing frames for 1000ms with 180fps requested
180 frames were captured at 180fps
frame delta time[us] distribution
      1 
      6 5544
     92 5545
     76 5546
      5 5547
after skip frame indices (middle column)
0% frame skips
pi@muska:~/raspiraw/tools $ ./raw2ogg2anim hd0.32k 0 10 180 file.ogg
removing old auxiliary files
copying /dev/shm/out.????.raw files
cat: /dev/shm/out.0000.raw: No such file or directory
10     
dcraw each .raw file (to .ppm)
Cannot decode file out.0000.raw
Cannot decode file out.0001.raw
Cannot decode file out.0002.raw
Cannot decode file out.0003.raw
Cannot decode file out.0004.raw
Cannot decode file out.0005.raw
Cannot decode file out.0006.raw
Cannot decode file out.0007.raw
Cannot decode file out.0008.raw
Cannot decode file out.0009.raw
Cannot decode file out.0010.raw
out.0010.raw     
.ppm -> .ppm.d
double: double.c:11: main: Assertion `src' failed.
./raw2ogg2anim: line 28:  5407 Aborted                 double $f > $f.D
double: double.c:12: main: Assertion `0 == strcmp(line, "P6\n") || 0 == strcmp(line, "P5\n")' failed.
./raw2ogg2anim: line 28:  5408 Aborted                 double $f.D > $f.d
out.*.ppm     
.ppm.d -> .ppm.d.png
./raw2ogg2anim: line 47: pnmtopng: command not found
out.*.ppm.d     
now creating hd0.32k.ogg
./raw2ogg2anim: line 53: gst-launch-1.0: command not found
now creating hd0.32k.anim.gif
/home/pi/raspiraw/tools/gifenc.sh: 11: /home/pi/raspiraw/tools/gifenc.sh: ffmpeg: not found
/home/pi/raspiraw/tools/gifenc.sh: 12: /home/pi/raspiraw/tools/gifenc.sh: ffmpeg: not found
``

May I kindly get your help when you get a chance to reply?

Thanks!
Kaan

6by9, 2018-05-15:
All those scripts have been written by @Hermann-SW.

The first issue appears to be that you haven't installed half the required tools
```sudo apt install netpbm ffmpeg gstreamer-tools``` may help for part of it. I haven't looked into which bits of GStreamer are being used, so you may need various additional packages there.

Hermann-SW, 2018-05-15:
I agree with 6by9 that needed tools have not been installed (ffmpeg).

The output above is badly readable, should have pasted into a code block.
I reformatted and the capturing was successful (180 frames captured in 1s, no frame skips):
`pi@muska:&#126;/raspiraw/tools $ ./320x240 1000
removing /dev/shm/out.*.raw
capturing frames for 1000ms with 180fps requested
180 frames were captured at 180fps
frame delta time[us] distribution
1 
6 5544 
92 5545 
76 5546 
5 5547
after skip frame indices (middle column)
0% frame skips
pi@muska:~/raspiraw/tools $

`

Readme.md clearly mentions that gstreamer is used:
[https://github.com/6by9/raspiraw](https://github.com/6by9/raspiraw)

In addition documentation on raw2ogg2anim there mentions ffmpeg and gifenc.sh as needed.

kunguz, 2018-05-15:
Here is an update after installing ffmpeg, gstreamer-tools, netpbm. I can now see some more lines printed to the shell during a capture process, but result is more or less the same. The raw files aren't decoded.
``
pi@muska:~/raspiraw/tools $ ./320x240 1000
removing /dev/shm/out.*.raw
capturing frames for 1000ms with 180fps requested
ls: cannot access '/dev/shm/out.*.raw': No such file or directory
0 frames were captured at 332fps
frame delta time[us] distribution
      1 
      6 3003
      8 3004
    137 3005
    166 3006
      6 3007
      8 3008
after skip frame indices (middle column)
3006,2,1333811159
3005,3,1333814164
3005,4,1333817169
3006,5,1333820175
3008,6,1333823183
3003,7,1333826186
3006,8,1333829192
3005,9,1333832197
3006,10,1333835203
3007,11,1333838210
3004,12,1333841214
3005,13,1333844219
3006,14,1333847225
3005,15,1333850230
3006,16,1333853236
3006,17,1333856242
3005,18,1333859247
3006,19,1333862253
3005,20,1333865258
3006,21,1333868264
3005,22,1333871269
3006,23,1333874275
3005,24,1333877280
3006,25,1333880286
3006,26,1333883292
3005,27,1333886297
3006,28,1333889303
3005,29,1333892308
3006,30,1333895314
3005,31,1333898319
3008,32,1333901327
3003,33,1333904330
3006,34,1333907336
3005,35,1333910341
3006,36,1333913347
3006,37,1333916353
3005,38,1333919358
3008,39,1333922366
3004,40,1333925370
3005,41,1333928375
3005,42,1333931380
3006,43,1333934386
3006,44,1333937392
3005,45,1333940397
3005,46,1333943402
3006,47,1333946408
3006,48,1333949414
3005,49,1333952419
3006,50,1333955425
3005,51,1333958430
3006,52,1333961436
3005,53,1333964441
3006,54,1333967447
3006,55,1333970453
[... 298 lines cut ...]
Cannot decode file out.0004.raw
Cannot decode file out.0005.raw
Cannot decode file out.0006.raw
Cannot decode file out.0007.raw
Cannot decode file out.0008.raw
Cannot decode file out.0009.raw
Cannot decode file out.0010.raw
out.0010.raw     
.ppm -> .ppm.d
double: double.c:11: main: Assertion `src' failed.
./raw2ogg2anim: line 28:  8921 Aborted                 double $f > $f.D
double: double.c:12: main: Assertion `0 == strcmp(line, "P6\n") || 0 == strcmp(line, "P5\n")' failed.
./raw2ogg2anim: line 28:  8922 Aborted                 double $f.D > $f.d
out.*.ppm     
.ppm.d -> .ppm.d.png
pnmtopng: EOF / read error reading magic number
out.*.ppm.d     
now creating hd0.32k.ogg
./raw2ogg2anim: line 53: gst-launch-1.0: command not found
now creating hd0.32k.anim.gif
hd0.32k.ogg: No such file or directory
hd0.32k.ogg: No such file or directory
done

``

Hermann-SW, 2018-05-15:
Not sure what you did. As I commented you successfully captured 180 frames the last time. This time you captured 0 frames. Most likely you executed raspivid or raspistill before, or restarted. In that case you need to cd into raspiraw folder and run ./camera_i2c.

You can avoid strikethrough in code by replacing "~" with "&#126;", tilde has special meaning in github.

This commandline is wrong as well:
`./raw2ogg2anim hd0.32k 0 10 180 file.ogg`

Tool tells you how to call
`$ raw2ogg2anim 
format: raw2ogg2anim vname first last fps [d[d]]
$ 

`

You should not overwrite the needed file hd0.32k.
This will create tst.ogg and tst.anim.gif, containg frames 1 to 10, playing at 180fps:
`./raw2ogg2anim tst 1 10 180`

Because of the temporary files created you should better work in a directory different to tools, tools directory need to be on $PATH.

Hermann-SW, 2018-05-15:
What is size of hd0.32k for you? It should be 32KB:
`$ ls -l hd0.32k 
-rw-r--r-- 1 pi pi 32768 May 12 23:30 hd0.32k
$ 
`

Hermann-SW, 2018-05-15:
From Readme.md -- you likely have not 6by9's version of dcraw:
The raw Bayer format frames captured by raspiraw can be converted to .ppm images by modified dcraw image processing app: https://github.com/6by9/dcraw

kunguz, 2018-05-15:
I have added the tools to the bashrc,  and I have also cloned dcraw of 6by9 and run buildme right after installing dependencies

```
pi@muska:~/dcraw $ sudo apt-get install libjasper-dev libjpeg8-dev gettext liblcms2-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libjasper1 libjpeg8
Suggested packages:
  gettext-doc autopoint libasprintf-dev libgettextpo-dev libjasper-runtime
The following NEW packages will be installed:
  gettext libjasper-dev libjasper1 libjpeg8 libjpeg8-dev liblcms2-dev
0 upgraded, 6 newly installed, 0 to remove and 0 not upgraded.
Need to get 7,224 kB of archives.
After this operation, 13.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirror.us.leaseweb.net/raspbian/raspbian stretch/main armhf libjasper1 armhf 1.900.1-debian1-2.4+deb8u1 [110 kB]           
Get:2 http://mirrors.syringanetworks.net/raspbian/raspbian stretch/main armhf libjpeg8 armhf 8d1-2 [108 kB]                             
Get:3 http://raspbian.mirror.constant.com/raspbian stretch/main armhf gettext armhf 0.19.8.1-2 [1,434 kB]
Get:4 http://raspbian-us.ngc292.space/raspbian stretch/main armhf libjasper-dev armhf 1.900.1-debian1-2.4+deb8u1 [501 kB]                                    
Get:5 http://mirrors.syringanetworks.net/raspbian/raspbian stretch/main armhf libjpeg8-dev armhf 8d1-2 [205 kB]                              
Get:6 http://raspbian-us.ngc292.space/raspbian stretch/main armhf liblcms2-dev armhf 2.8-4 [4,866 kB]                                        
Fetched 7,224 kB in 6s (1,124 kB/s)                                                                                                                                                                                                           
Selecting previously unselected package libjasper1:armhf.
(Reading database ... 124556 files and directories currently installed.)
Preparing to unpack .../0-libjasper1_1.900.1-debian1-2.4+deb8u1_armhf.deb ...
Unpacking libjasper1:armhf (1.900.1-debian1-2.4+deb8u1) ...
Selecting previously unselected package libjpeg8:armhf.
Preparing to unpack .../1-libjpeg8_8d1-2_armhf.deb ...
Unpacking libjpeg8:armhf (8d1-2) ...
Selecting previously unselected package gettext.
Preparing to unpack .../2-gettext_0.19.8.1-2_armhf.deb ...
Unpacking gettext (0.19.8.1-2) ...
Selecting previously unselected package libjasper-dev.
Preparing to unpack .../3-libjasper-dev_1.900.1-debian1-2.4+deb8u1_armhf.deb ...
Unpacking libjasper-dev (1.900.1-debian1-2.4+deb8u1) ...
Selecting previously unselected package libjpeg8-dev:armhf.
Preparing to unpack .../4-libjpeg8-dev_8d1-2_armhf.deb ...
Unpacking libjpeg8-dev:armhf (8d1-2) ...
Selecting previously unselected package liblcms2-dev:armhf.
Preparing to unpack .../5-liblcms2-dev_2.8-4_armhf.deb ...
Unpacking liblcms2-dev:armhf (2.8-4) ...
Setting up liblcms2-dev:armhf (2.8-4) ...
Setting up libjasper1:armhf (1.900.1-debian1-2.4+deb8u1) ...
Setting up gettext (0.19.8.1-2) ...
Processing triggers for install-info (6.3.0.dfsg.1-1+b1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up libjpeg8:armhf (8d1-2) ...
Setting up libjasper-dev (1.900.1-debian1-2.4+deb8u1) ...
Setting up libjpeg8-dev:armhf (8d1-2) ...
pi@muska:~/dcraw $ ./buildme 
/usr/bin/ld: warning: libjpeg.so.62, needed by /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libjasper.so, may conflict with libjpeg.so.8
```

and proceeded with the capture as in below:

```
pi@muska:~ $ camera_i2c 
setting GPIO for board revsion: 9000c1
PiZero W - I2C 0 on GPIO 28 & 29. GPIOs 40 & 44 for LED and power
pi@muska:~ $ raspiraw -md 7 -t 1000 -ts tstamps.csv -hd0 hd0.32k -h 64 --vinc 1F --fps 660 -r "380A,0040;3802,78;3806,0603" -sr 1 -o /dev/shm/out.%04d.raw 2>/dev/null
Using i2C device /dev/i2c-0
pi@muska:~ $ ls -l /dev/shm/out.*.raw | wc --lines
332
pi@muska:~ $ cat hd0.32k /dev/shm/out.0123.raw > out.0123.raw
pi@muska:~ $ cd dcraw/
pi@muska:~/dcraw $ ./dcraw ~/out.0123.raw 
pi@muska:~/dcraw $ double out.0123.ppm > out.0123.ppm.d
double: double.c:11: main: Assertion `src' failed.
Aborted
```

Hermann-SW, 2018-05-15:
Ok, all sems to have worked for you.
Now you need to bring in bright light, otherwise you will see ferames like this.
You captured with 660fps, which leaves 1.6ms time for frames.
I used 1000lm light successfully up to 665fps.
For v2 camera 1000fps recording I have to use 5000lm very bright light, see:
[https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=212518&p=1310445#p1310442](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=212518&p=1310445#p1310442)
My standard test object are some lego pieces with yellow, red, green, blue, white and block colors

kunguz, 2018-05-15:
Got it, thanks for all the help. I will try the bright light addition when I get back to my home as I am doing all these on a remote raspberry pi at home.

6by9, 2018-06-26:
Closing issue as resolved.

closed 2018-06-26 by 6by9