raspberrypi/raspiraw #16: Compile error

lemarquois, 2018-08-17:
Hi,

I cloned this repository directly on a raspberry pi 3 b+ and tried to compile the sources without success:
pi@raspberrypi:~/raspiraw $ ./buildme
In file included from /opt/vc/include/interface/vcos/vcos_assert.h:149:0,
                 from /opt/vc/include/interface/vcos/vcos.h:114,
                 from raspiraw.c:42:
/opt/vc/include/interface/vcos/vcos_types.h:38:33: fatal error: vcos_platform_types.h: No such file or directory
 #include "vcos_platform_types.h"
                                 ^
compilation terminated.
In file included from /opt/vc/include/interface/vcos/vcos_assert.h:149:0,
                 from /opt/vc/include/interface/vcos/vcos.h:114,
                 from RaspiCLI.c:46:
/opt/vc/include/interface/vcos/vcos_types.h:38:33: fatal error: vcos_platform_types.h: No such file or directory
 #include "vcos_platform_types.h"
                                 ^
compilation terminated.

There is however a file called vcos_platform_types.h in my filesystem it is located at:
/opt/vc/include/interface/vcos/pthreads/vcos_platform_types.h

Any hints?

Thanks

lemarquois, 2018-08-17:
pi@raspberrypi:~/raspiraw $ make
gcc -I/opt/vc/include -pipe -W -Wall -Wextra -g -O2 -c -o raspiraw.o raspiraw.c
In file included from /opt/vc/include/interface/vcos/vcos_assert.h:149:0,
                 from /opt/vc/include/interface/vcos/vcos.h:114,
                 from raspiraw.c:42:
/opt/vc/include/interface/vcos/vcos_types.h:38:33: fatal error: vcos_platform_types.h: No such file or directory
 #include "vcos_platform_types.h"
                                 ^
compilation terminated.
Makefile:9: recipe for target 'raspiraw.o' failed
make: *** [raspiraw.o] Error 1

lemarquois, 2018-08-17:
okay, this is very strange. I also had a Raspbian Stretch Lite (2018-04-18-raspbian-stretch-lite.zip) and I did not have the lines you posted above in my include file.
I now downloaded the newest version 2018-06-27-raspbian-stretch-lite.zip and set up everything from scratch. It complies now as expected. Both commands work for me, the CSI camera already shows up on the I2C (./camera_i2c).

I'm trying to interface a CSI-Camera that is currently not in your sources. If I encounter any problems, shall I post them here as issues or on the raspberry pi forum?

Best regards,

Markus

closed 2018-08-17 by lemarquois

6by9, 2018-08-17:
How old was your install? Those lines were added in https://github.com/raspberrypi/userland/commit/826b0b1b089329cdd9c3c9a3a91c0ffe915cf248#diff-8f8e8ef15955ad7af43eb9a3cbefa22b in March 2016, so an old Jessie install may still have had the old userland.

Please post issues in interfacing your camera on the forum.
Out of interest, what is the device?