raspberrypi/raspiraw #19: ov5647 camera not detected

cfsterpka, 2018-10-05:
I'm having difficulties detecting my camera.  The output of the commands is: 

$ ./camera_i2c 
setting GPIO for board revsion: 900093
A+, B+, B2, B2 (2837), and Zero (not W) all revisions - I2C 0 on GPIOs 28 & 29. GPIOs 32 & 41 for LED and power

$ ./raspiraw -md 7 -t 1000 -ts tstamps.csv -hd0 hd0.32k -h 64 --vinc 1F -y 1 --fps 660 -r "380A,0040;3802,78;3806,0603" -sr 1 -o /dev/shm/out.%04d.raw 
Using i2C device /dev/i2c-1
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: No sensor found. Aborting

$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

I have the line: "dtparam=i2c_arm=on" in my config.txt file, and the camera is enabled as well (I can take pictures with raspistill).  I'm not sure what I'm missing!

6by9, 2018-10-05:
On a Pi 0 I'd be expecting you to have added ```dtparam=i2c_vc=on``` and not be using ```-y 1``` in your raspiraw command-line.
Where have you found comment that you should be using i2c-1?

cfsterpka, 2018-10-05:
When I enable i2c on my Pi it automatically adds "dtparam=i2c_arm=on" and not "dtparam=i2c_vc=on".  I used "-y 1" because the only device created is /dev/i2c-1.  I tried using "dtparam=i2c_vc=on", but no i2c devices are created when I do so.

6by9, 2018-10-05:
https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=109137
>  You do need to jump through the hoops to enable /dev/i2c-0 first (see the "Interfacing" forum, but it should just be adding "dtparam=i2c_vc=on" to /boot/config.txt, and "sudo modprobe i2c-dev").

cfsterpka, 2018-10-05:
Awesome - that worked thanks!

closed 2018-10-05 by cfsterpka