raspberrypi/raspiraw #22: Colors

kozuch, 2018-10-23:
This is more of a question. The colors for these high FPS images are not what they should be I see (at least for OV5647). What happens there? I guess some noise plays in? Also if lines are skipped the bayer demosaicing probably does not fully work?

Also, why is the finger half-transparent on one of the GIF examples?

Hermann-SW, 2018-10-30:
On the colors I have to say that some cobinations work, and some don't. The tools in tools directory give "correct" colors. WHat you need to know is that for very high framerates (>400fps) you need very bright light to compensate the very small shutter times. I normally used a 1000lm Ikea lamp, but for 1007fps videos a 5000lm light was needed.

> Also, why is the finger half-transparent on one of the GIF examples?
>
I have no idea, must be a bad setting of increments used at that time. Again, raspiraw used with the sesstings from tools directory tools should work fine.

You can find all information I found under this bookmark list:
[https://stamm-wilbrandt.de/en/Raspberry_camera.html#raspiraw](https://stamm-wilbrandt.de/en/Raspberry_camera.html#raspiraw)

v1 camera was easier, the most difficult part to find out was how to do v2 camera raspiraw recording right with line skipping. Tool 640x150_s records 640x150@10007fps, but records only every other line. Postprocessing needs to duplicate each line to result in 640x150 frame.

6by9, 2019-01-23:
Closing as this is really a request over register settings, which I can't discuss due to NDAs.

closed 2019-01-23 by 6by9