raspberrypi/raspiraw #24: 12c struct conflict

tvjon, 2019-01-23:
Just reading the RPi forum post:
https://www.raspberrypi.org/forums/viewtopic.php?t=231937&p=1420408#p1420408

I too was unaware of your "tools" & it looks interesting, thank you.

However, I'm unable to build.

$ make
gcc -I/opt/vc/include -pipe -W -Wall -Wextra -g -O0 -c -o raspiraw.o raspiraw.c
In file included from raspiraw.c:40:0:
/usr/include/linux/i2c-dev.h:37:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^~~~~~~
In file included from raspiraw.c:39:0:
/usr/include/linux/i2c.h:68:8: note: originally defined here
 struct i2c_msg {
        ^~~~~~~
In file included from raspiraw.c:40:0:
/usr/include/linux/i2c-dev.h:89:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~
In file included from raspiraw.c:39:0:
/usr/include/linux/i2c.h:131:7: note: originally defined here
 union i2c_smbus_data {
       ^~~~~~~~~~~~~~

Is this perhaps a result of moving to the .19 kernel?

6by9, 2019-01-23:
https://github.com/6by9/raspiraw/blob/master/buildme
Seems fine on my Raspbian install on 4.19.

tvjon, 2019-01-23:
Odd,
your command line gives the same errors.

Ok, i'll try to look at it in detail tonight, thank you.

6by9, 2019-01-23:
I don't know where you got your kernel headers from, but 4.19 looks sane.
```struct i2c-msg``` defined at i2c.h lnie 69 https://elixir.bootlin.com/linux/v4.19.17/source/include/uapi/linux/i2c.h#L69
i2c-dev.h doesn't redefine it, https://elixir.bootlin.com/linux/v4.19.17/source/include/uapi/linux/i2c-dev.h#L38, and it doesn't have a line 89.

tvjon, 2019-01-23:
Ok, now built.

I commented out L40
//#include <linux/i2c-dev.h>

After L294 void update_regs(...

I added:

/* This is the structure as used in the I2C_RDWR ioctl call */
struct i2c_rdwr_ioctl_data {
	struct i2c_msg *msgs;	/* pointers to i2c_msgs */
	__u32 nmsgs;			/* number of i2c_msgs */
};

#define I2C_RDWR	0x0707	/* Combined R/W transfer (one STOP only) */

all taken from the nop'd

/usr/include/linux/i2c-dev.h

Just on the off chance that it's relevant, obviously for several reasons you cross compile, but I always build on a RPi, & in fact have no facility to cross compile.

As always thank you for your time looking at this.

tvjon, 2019-01-23:
Ah, the web page idn't update until I posted the above, so I hadn't seen your last message.

6by9, 2019-01-23:
I don't cross-compile this app as it is so trivial, so I'm building natively too.

tvjon, 2019-01-24:
Ok.
Thank you for those links

closed 2019-01-24 by tvjon