raspberrypi/raspiraw #25: raspiraw "--expus" does not work for ov5647 (it does work for imx219)

Hermann-SW, 2019-03-03:
Doing raspiraw captures with imx219 for "--expus 1000" and "--expus 2000" gives different bright frames after processing with dcraw.

Doing the same with ov5647 give identical frames, no difference in brightness.

I tested that this bug exists on this repo as well.

6by9, 2019-03-03:
See #18.
Getting a calculation that can be parameterised and is correct for both sensors is trickier than it looks.

Hermann-SW, 2019-03-04:
I used your latest raspiraw and setting shutter does not work at all.
I cannot find a repo of gustav-engdahl but only see his commits on your branch.
But the pull request is anot integrated into your branch yet.
How can I test whether the three commits in that pull request make "-e" work on your repo?
(without copying over line by line by hand)

GustavEngdahl, 2019-03-04:
Hi,

You should be able to build my repo at:
https://github.com/GustavEngdahl/raspiraw/commits/master
It's master has the three patches needed to get the VTS/Exposure/gain
values correctly written. The patches are not chip specific in any way, but
I've only verified on OV5647.

Regards,
Gustav

Hermann-SW, 2019-03-04:
Thank you, I did build your repo and it fixes the issue I reported here, closing this issue as duplicate now.
These are the (different now) frames with different --expus settings.

I did not look in detail into your changes, but "--expus" seems to have worked because for imx219 because exposure bits are a multiple of 8 whereas for ov5647 they are not:

```
      .exposure_reg =         0x015A,
      .exposure_reg_num_bits = 16,

   .exposure_reg =         0x3500,
   .exposure_reg_num_bits = 20,
```

[--expus 4000](https://stamm-wilbrandt.de/en/forum/72kmh/out.0017.ppm.d.png)
[--expus 2000](https://stamm-wilbrandt.de/en/forum/72kmh/out.0016.ppm.d.png)
[--expus 1000](https://stamm-wilbrandt.de/en/forum/72kmh/out.0015.ppm.d.png)
[--expus 500](https://stamm-wilbrandt.de/en/forum/72kmh/out.0014.ppm.d.png)

closed 2019-03-04 by Hermann-SW

GustavEngdahl, 2019-03-04:
Good to hear.

Can you verify that my fixes doesn't break imx219? If nothing breaks,
perhaps 6by9 could bring in #18 to master?

Regards,
Gustav