raspberrypi/raspiraw #27: PI3A+ ov5647 not detected

CZLYddy, 2019-05-10:
Hi,master!
 My PI3A+ run ./camera_i2c,output:
setting GPIO for board revsion: 9020e0
Failed: don't know how to set GPIO for this board!
Can you help me?

6by9, 2019-05-10:
camera_i2c didn't know the revision code, exactly as it said.
I've just pushed a commit that should match the required configuration, but I haven't got a 3A+ to test.

CZLYddy, 2019-05-10:
Thank you very much!
$ ./camera_i2c
setting GPIO for board revsion: 9020e0
Raspberry Pi3B/Pi3B+/3A
Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- 36 -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

but there is a question:

$ ./raspiraw -md 7 -t 1000 -ts tstamps.csv -hd0 hd0.32k -h 64 --vinc 1F --fps 660 -r "380A,0040;3802,78;3806,0603" -sr 1 -o /dev/shm/out.%04d.raw 2>/dev/null
Using i2C device /dev/i2c-0
No AWB
$ ls -l /dev/shm/out.*.raw | wc --lines
210

why is the fps  210 but not 660?

6by9, 2019-05-10:
(a) I don't support the reduced resolution modes, or indeed any of the lower level tinkering with the I2C register set, due to having information under NDA.

(b) Are you sure your /dev/shm drive is large enough to store 660 frames of that size?

Original issue resolved so I'm closing this.

closed 2019-05-10 by 6by9

CZLYddy, 2019-05-10:
Thank you so much!