raspberrypi/raspiraw #30: Raspiraw + dcraw results in gibberish images

locinus, 2019-10-21:
Hi! Thanks for your fine libraries!
Everything was working nicely until we lost our Pi SD-card (with Jessie) and configured a new one (with Buster). I've been trying to understand what changed and to tweak configurations but I can't have it to work again as before.

Now raspiraw + dcraw provide us with gibberish pictures such as this one: 
[image]

**Configuration:**

- Raspberry PI3 B+
- Linux demo 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l
- Raspbian GNU/Linux 10

We use a multi-camera adapter (Arducam) to communicate with 3 different cameras. To switch from one camera to another, you use this snippet:
```
import RPi.GPIO as gp
gp.setwarnings(False)
gp.setmode(gp.BOARD)
gp.setup(7, gp.OUT)
gp.setup(11, gp.OUT)
gp.setup(12, gp.OUT)
# init on camera 1
gp.output(7, False)
gp.output(11, False)
gp.output(12, True)        
```

**Now, some investigation clues:**

- **the cameras do work and are reachable**; we can stream them using picamera, with appropriate image results, using this code https://gist.github.com/locinus/f50ee64734a0eda2bdcedea6f098e01e

- each camera provides a gibberish image of its own, some a bit darker from others; AND each new shot from a given set camera is very slighty different from the last

Examples for camera 1:
[image]
[image]

Examples for camera 2:
[image]
[image]

Examples for camera 3:
[image]
[image]

=>this seems to indicate that the data from each camera is actually used, only it's not rendered well!

- in our previous system, we **could** use the streaming with picamera and then take raw pictures with raspiraw (not both at the same time though)

Lastly, another problem (which might be related or not) which appeared when we reinstalled the system:
now, when we stream the camera's video, it breaks whatever configuration camera_i2c sets (just calling picamera.PiCamera(resolution='640x480', framerate=24) is enough to break it), and raspiraw then returns:
```
Using i2C device /dev/i2c-0
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: No sensor found. Aborting

```
which a call to camera_i2c fixes, but it might indicate a deeper issue.

6by9, 2019-10-21:
Where have you got your copy of dcraw from? The stock dcraw supports one particular format of raw from the OV5647. The version in https://github.com/6by9/dcraw supports all the Pi camera modules by parsing the raw header.

The firmware does reset the GPIOs for the camera when it closes the sensor. That's always been the case, and is one of the reasons that it is not recommended to mix raspivid/raspistill with raspiraw.

locinus, 2019-10-21:
Oh me !
**So I resolved the main issue.**
The issue being that our legacy code was actually **using the -eus parameter, with an absolutely wrong value (1)**. With such a low exposure time, the picture would show almost nothing but noise.
_Yet, it seemed to work in our previous setting but not now...?_

The only question that remains is: why does the streaming of the video (with picamera.PiCamera) breaks what camera_i2c sets? Indeed, now we have to systematically call camera_i2c after the streaming, when before it was not necessary.

6by9, 2019-10-21:
> The issue being that our legacy code was actually using the -eus parameter, with an absolutely wrong value (1). With such a low exposure time, the picture would show almost nothing but noise.
> Yet, it seemed to work in our previous setting but not now...?

There's an outstanding issue #20 to actually make the I2C register programming generic. 
https://github.com/6by9/raspiraw/commit/fbe2936d1ab468f6b1367bbd0a6188f247d96189 was commited which fixed it for one sensor (probably imx219)

>  Indeed, now we have to systematically call camera_i2c after the streaming, when before it was not necessary.

It's always been necessary.
The firmware powers down the camera after every use by clearing the shutdown (and LED) GPIOs.  The only way to stop that would be to make a dt-blob.bin that removed the defines so the firmware never tries to power the sensor up or down, but I wouldn't recommend that.

locinus, 2019-10-21:
Indeed you're right. I found in the legacy code a part where camera_i2c is called before every snapshot.

That closes it. Thank you very much for your time and support !

closed 2019-10-21 by locinus