raspberrypi/raspiraw #31: why don't use rawcam output yuv

TongleiJin, 2019-11-27:
Now the project output yuv by isp component, why not use rawcam output yuv directly? I did see "MMAL_ENCODING_I420" likes in "mmal_encodeings.h".

6by9, 2019-11-27:
The rawcam component outputs whatever format the CSI2 source device generates, be that Bayer, RGB, YUYV, or something else.
Other than potentially repacking the Bayer formats, there is no image processing applied.

CSI2 devices are almost always line based to avoid needing full frame buffers, therefore planar formats such as I420 would be very unusual.
If you read the CSI2 spec for the supported YUV4:2:0 formats, they specify it as an interleaved format with either:
- odd lines being UYYUYYUYY, and even lines being VYYVYYVYY
- odd lines being half the length of even lines and only having luma (Y) samples, even lines being UYVYUYVY (same as one of the YUV4:2:2 interleavings).

I don't think I've ever come across hardware which can consume that!

The ISP is designed to accept the majority of these CSI2 formats and convert them into the formats that are more commonly used within computer systems, eg planar YUV, or RGB.
(The ISP is the only hardware block within VideoCore that handles YUYV interleaved YUV4:2:2 for example)

6by9, 2020-02-25:
I believe this question has been answered, so closing.
Please reopen if you wish to continue the discussion.

closed 2020-02-25 by 6by9