raspberrypi/raspiraw #33: rpi zero and rpi camera v2 imx219, no output

sylvainblot, 2019-12-18:
Hello,

I can't get a single frame to be captured, the output file is always empty, and there is a registry that can't be set. I can see the image using the preview mode.

```./raspiraw -md 1 --height 1080 --top 0 --fps 30 -sr 1 -o out.raw
Using i2C device /dev/i2c-0
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Found sensor imx219 at address 10
RaspiRaw: Reg: 380A not found!

RaspiRaw: Reg: 380A not found!

RaspiRaw: Reg: 380A not found!

RaspiRaw: Reg: 380A not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Reg: 380B not found!

RaspiRaw: Encoding 41414270
No AWB
mmal: Set pack to 0, unpack to 0
mmal: Timing 6/2, 2/6/0, 0/0
mmal: Create pool of 6 buffers of size 2611200
mmal: Create pool of 6 buffers of size 2611200
mmal: Now streaming...

 ls -lah out.raw 
-rw-r--r-- 1 root root 0 Feb 14 10:23 out.raw
```

Regards

6by9, 2019-12-18:
Does the camera module work with raspivid/still? If not then look for the normal issues around partially seated connectors.

You're using essentially defaults anyway, so cut the command line down to ```./raspiraw -md 1 -o out.raw```

sylvainblot, 2019-12-18:
my bad wrong timeout I can capture a single frame.
Sorry

closed 2019-12-18 by sylvainblot

sylvainblot, 2019-12-19:
@6by9 thanks, just saw your comment, is there a way to convert the raw output to dng ? I need that to correct colors.  Thanks

sylvainblot, 2019-12-19:
(and use the images in an other software)

6by9, 2019-12-19:
https://github.com/illes/raspiraw, or more sensibly the forks at https://github.com/bablokb/raspiraw or https://github.com/dword1511/raspiraw which have been updated for imx219.
The original thread discussing the raw format and developing that app is at https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=44918