raspberrypi/raspiraw #50: Raspiraw reaches 'No AWB' but does finish command (Raspberry Pi 4 B, IMX477)

Emmett-Dec, 2021-02-18:
I am running buster and my setup includes a Pi 4 model B and the HQ camera sensor.
[image]

rolly-ng, 2021-02-20:
-y 10 means that you are calling i2c address 10, whereas raspiraw found your camera on i2c-1a?

6by9, 2021-02-20:
> -y 10 means that you are calling i2c address 10, whereas raspiraw found your camera on i2c-1a?

No, -y 10 means use /dev/i2c-10 to look for sensors as you can see from the following line
"Using I2C device /dev/i2c-10"
This is correct for any 5.4 or later Pi kernel.
Each sensor has a hard coded address at which a specific register is checked to see if it matches the hard coded "signature" for that sensor.

@Emmett-Dec does this HQ camera work OK with raspistill/vid? Do modes other than -md 3 work?

Emmett-Dec, 2021-02-20:
Yes, raspistill/vid work perfectly with the HQ cam.

Just ran the same command this morning after restarting my Pi (with -md 3) and the command finished but the .raw was empty
[image]
[image]

Switching to mode 2 worked and resulted in a 3.2 MB .raw file
Mode 1 worked and resulted in a 4.56 MB .raw file
Mode 0 resulted in this error:
[image]

rolly-ng, 2021-02-21:
Thanks to 6by9 for the correction. Plz try to increase your GPU memory to 256MB at least.

Emmett-Dec, 2021-02-21:
`uname -a` shows
`Linux GillesPi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux`

I've set GPU memory to 512 as suggested

This is what my GPIO looks like when running `camera_i2c` before restarting
[image]

I then reset my Pi and `camera_i2c` did the following:
[image]

I've noticed that raspiraw seems to work for my setup when a:10 is empty and 4:60 is set to 64, but `camera_i2c` sets the GPIO to either of those two states. Can anyone comment on this?

6by9, 2021-02-21:
The power to the camera module (on address 0x1a) is controlled via a GPIO.
The firmware camera driver (ie raspistill/vid) will set and clear that GPIO as the camera is used.
camera_i2c will set the GPIO to enable the camera, and it will be left enabled, hence address 0x1a responding via i2cdetect.

The result of the i2cdetect from within camera_i2c means relatively little - the sensor takes about 250ms to power up fully after power is applied, and camera_i2c doesn't wait at all before probing.

rolly-ng, 2021-02-23:
> `uname -a` shows
> `Linux GillesPi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux`
>
> I've set GPU memory to 512 as suggested
>
> This is what my GPIO looks like when running [...]

I am have board revision d03114 and I usually repeat ./camera_i2c 1a twice to have the a10 showing 1a before running raspiraw. It works fine so far.

Emmett-Dec, 2021-02-25:
The increase in memory seems to have fixed the earlier issue. I am now getting issues with all data being written to a single .raw file when using `./raspiraw -md 2 -t 1000 -ts /dev/shm/tstamps.csv -hd0 /dev/shm/hd0.32k --fps 660 -sr 1 -o /dev/shm/out.%06d.raw`
I am having the same issue when lowering fps to 350.
[image]

closed 2021-03-11 by Emmett-Dec