raspberrypi/raspiraw #55: Can't get it to work with P4B board revision c03112

Mut1nyJD, 2021-04-14:
Hi

It seems I am unable to get raspiraw to work on my P4B and the PiCamera V2. 
When I execute camera_i2c I'll get the following output:

setting GPIO for board revsion: c03112
4B(1G/2G/4G/8G)
Set state of 133 to 1
Use i2c-10 for the sensor (-y 10)

When I then try running ./raspiraw -md 1 -y10 

Using I2C device /dev/i2c-10
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: Probing sensor imx477 on addr 1A
RaspiRaw: No sensor found. Aborting

raspistill/raspivid both work fine

This is the version of the OS I am running
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux

I am using the head of the master-branch at the moment.

Any thoughts on what could be the problem here? I've seen other people getting it to run on their P4B

6by9, 2021-04-14:
Run ```./rpi3-gpiovirtbuf s 133 1``` and then ```sudo i2cdetect -y 10```. Does i2cdetect report any detected addresses?

Update your system as 5.10 is now the default.

Mut1nyJD, 2021-04-14:
@6by9 

Thank you for the quick answer, hmm well one of the issues could be that there is no /dev/i2c-10 at all there is only /dev/i2c-1

Mut1nyJD, 2021-04-14:
forgot to add that i2cdetect -y 10 of course does not work because of the missing /dev/i2c-10

6by9, 2021-04-14:
```dtparam=i2c_vc=on``` to /boot/config.txt

margaritaetchegaray, 2021-04-15:
Hi :) I have a similar problem. I have a P4B and the HQ camera V1. 
When I run the camera_i2c command I get a similar response to Mut1nyJD. Here the picture of my terminal (I also tried what you proposed to Mut1nyJD).

I have already done these steps:
dtparam=i2c_vc=on on the file /boot/config.txt

PATH=~/raspiraw:~/raspiraw/tools:$PATH on the file /.bashrc (I literally just copied the exact line, do I need to replace something?)

i2c-dev on the file /etc/modules-load.d/modules.conf

I ran a test command but it seems that I do not have the files on /dev/shm/out.*.raw 

I would be extremely grateful if you could provide any tips :)

[image]
[image]

6by9, 2021-04-15:
@margaritaetchegaray Totally different issue as you're not seeing 
```
RaspiRaw: No sensor found. Aborting
```
so it should have been a new issue avoid talking at cross-purposes.

You appear to have copied a command line for the V1 sensor and assumed it'll work for HQ camera.
The text in the [readme](https://github.com/raspberrypi/raspiraw#examples) even reads
> This command captures video from ov5647 camera on CSI-2 interface:

You're not using an ov5647 so it's not expected to work.

Mut1nyJD, 2021-04-21:
> `dtparam=i2c_vc=on` to /boot/config.txt

That did the trick thank you working now!

closed 2021-04-21 by Mut1nyJD