raspberrypi/raspiraw #56: Raspiraw for imx477 sensor

margaritaetchegaray, 2021-04-15:
In the raspiraw docuementation the list of supported sensors include: adv7282m, imx219, and ov5647.
I have a HQ camera and it seems like I have the imx477 sensor. Is there any way I can write the command for this sensor to create a video with approx. 600 fps?

6by9, 2021-04-15:
If others wish to work out the relevant register changes to achieve this then PRs are welcome.

Raspberry Pi have information from Sony under NDA, and are therefore are limited to what can be published without additional clearances from them.

kyleboddy, 2022-01-13:
I noticed imx477 files in the codebase along with others noting success. Is the imx477 sensor currently supported? Will file some tickets regarding my use of it (haven't had much success), but if still unsupported/doesn't work, don't want to fill the issue tracker with logs.