raspberrypi/raspiraw #59: raspiraw doesn't work with imx219 or ov5647

mmm545, 2021-09-11:
I followed the instructions to download raspiraw, i ran `make` and everything worked correctly, added `PATH=~/raspiraw:~/raspiraw/tools:$PATH` to .bashrc, closed the terminal and reopened it and then i ran camera_i2c and it gave me this output:
```
setting GPIO for board revsion: b03111
4B(1G/2G/4G/8G)
Set state of 133 to 1
Use i2c-0 for the sensor (-y 0)
```
I assumed everything is running correctly and started playing around with it.
First thing I did is trying to run this command:
```
raspiraw -md 7 -t 1000 -ts tstamps.csv -hd0 hd0.32k -h 64 --vinc 1F --fps 660 -r "380A,0040;3802,78;3806,0603" -sr 1 -o /dev/shm/out.%04d.raw 2>/dev/null
```
It printed out `Using I2C device /dev/i2c-10` and then immediately exited, running `ls -l /dev/shm/out.*.raw | wc --lines` gave me:
```
ls: cannot access '/dev/shm/out.*.raw': No such file or directory
0
```
Looking back at the i2c_camera output, it told me to use `-y 0`, so added that to the raspiraw command i ran above and this time it did use /dev/i2c-0, again it immediately exited after that, running the `ls` command gave me the same output.
A few issues i read told me to put `dtparam=i2c_vc=on` in config.txt so i did that and rebooted the raspberry pi.
After reboot i noticed that `/dev/i2c-0` does exist now, so i ran the exact same commands above  (with the -y 0 flag and also with the exception of `make` ofc), again same output.
So i switched to my other camera module that has the ov5647 sensor, *again* i ran camera_i2c, same output then i ran the raspiraw command above (with the -y 0 flag), this time however it printed out:
```
Using I2C device /dev/i2c-0
No AWB
```
one second later, the program exits.
Running the `ls` command printed `219`, however it doesn't seem to be the expected output since if it was told to record 660 frames then there should be 660 frames but nope, only 219.
Another issue suggested to raise the gpu_memory to 512mb, i did that and no difference at all.
I'm assuming that the raspiraw command above was specifically for the ov5647 sensor and that's why it gave a different output this time, but really i have no idea what i'm doing so ¯\\\_(ツ)_/¯
I tried running the 640x32 and 640x64 tools but both gave me this output:
```
removing /dev/shm/out.*.raw
capturing frames for 1000ms with 749fps requested
cut: tstamps.csv: No such file or directory
ls: cannot access '/dev/shm/out.*.raw': No such file or directory
/home/pi/raspiraw/tools/640x32: line 14: 1000000 / : syntax error: operand expected (error token is "/ ")
frame delta time[us] distribution
cut: tstamps.csv: No such file or directory
after skip frame indices (middle column)
grep: tstamps.csv: No such file or directory
grep: tstamps.csv: No such file or directory
wc: tstamps.csv: No such file or directory
expr: syntax error: unexpected ')'
% frame skips
```
with the 640x64 it records 659fps instead and seems that it pauses for a moment after `capturing frames for 1000ms with 659fps requested` but pretty much the same output.
I'm stuck here at this point with nothing working so any help is appreciated

6by9, 2021-09-11:
Redirecting stderr to /dev/null is not helpful when trying to debug things. Error messages are there for good reason.

You appear to have copied that command line from the README when the very next line reads
> This command captures video from **ov5647** camera on CSI-2 interface:

It is using the -r option to override register values. Register configuration is always sensor specific, therefore it will not apply to imx219 (or imx477). 

The use of /dev/i2c-0 implies that your system is not up to date. Raspberry Pi OS is using the 5.10 kernel which now exposes /dev/i2c-0 and /dev/i2c-10. Memory says the 5.4 kernel does the same. The 4.x kernels are pretty ancient.

The use of raspiraw, especially when messing with register settings, requires a moderate level of understanding of how the specific sensor works.

mmm545, 2021-09-11:
> It is using the -r option to override register values. Register configuration is always sensor specific, therefore it will not apply to imx219 (or imx477).

Yeah my bad, i knew that was the case but i only realized it when i started writing this issue, and i mentioned that above too :D
> The use of /dev/i2c-0 implies that your system is not up to date. Raspberry Pi OS is using the 5.10 kernel which now exposes /dev/i2c-0 and /dev/i2c-10. Memory says the 5.4 kernel does the same. The 4.x kernels are pretty ancient.

I will try to update the OS and see if it works.
> The use of raspiraw, especially when messing with register settings, requires a moderate level of understanding of how the specific sensor works.

I understand raspiraw goes into some low level stuff, but since there was an already written example i used it since i had no knowledge on how this stuff works (i shouldn't blindly work with this thing anyway)

Thanks.

closed 2021-09-11 by mmm545