raspberrypi/raspiraw #61: Acquistion interferences with dsi touchscreen

JDasHO, 2021-10-13:
Hi,

I am trying to use the Raspberry V2 camera (imx219) (connected to csi socket) together with the official raspberry touchscreen (connected to dsi socket) ([https://www.raspberrypi.com/products/raspberry-pi-touch-display/](url)). Doing so however generates several mmal errors (e.g. "mmal: Failed to write register index 3") as well as often 'disturbed' images. This goes for acquisitions both with or without preview.
The complete terminal output for an image acquisition looks like this:

> devicem6@devicem6:\~/raspiraw $ camera_i2c
>setting GPIO for board revsion: c03112
>4B(1G/2G/4G/8G)
>Set state of 133 to 1
>Use i2c-10 for the sensor (-y 10)
>devicem6@devicem6:\~/raspiraw $ raspiraw -o img.raw
>Using I2C device /dev/i2c-10
>RaspiRaw: Probing sensor ov5647 on addr 36
>RaspiRaw: Probing sensor imx219 on addr 10
>RaspiRaw: Found sensor imx219 at address 10
>RaspiRaw: Encoding 41414270
>No AWB
>mmal: Set pack to 0, unpack to 0
>mmal: Timing 6/2, 2/6/0, 0/0
>mmal: Create pool of 6 buffers of size 10171392
>mmal: Create pool of 6 buffers of size 10171392
>mmal: Failed to write register index 3
>mmal: Failed to write register index 9
>mmal: Failed to write register index 17
>mmal: Failed to write register index 29
>mmal: Failed to write register index 44
>mmal: Now streaming...
>devicem6@devicem6:\~/raspiraw $

The number of "Failed to write register..." errors does however vary from 2 to about 10 (in this case 5) with changing index numbers.
When I run the raspiraw image acquisition while the raspberry is connected to a normal desctop screen via HDMI instead of the touchscreen it works like a charm.
Further I tried to take images with raspistill, just for trouble shooting. Here I don't get any errors for both screen configurations.

Has anyone closer insights on what might be going wrong and how to fix it? I would apreciate it very much

6by9, 2021-10-13:
You have one I2C controller to which both the display/touchscreen and camera module are connected.
You have 2 processors - ARM and GPU (aka firmware). Only one of them can safely communicate with the above I2C block at a time. 

If you have the touchscreen enabled, then the GPU will be polling away reading it.
Using raspiraw which sends I2C to the camera from the ARM at that point is not safe, and will lead to conflicts in accessing the I2C block.

Disable the touchscreen element only by adding ```disable_touchscreen=1``` to /boot/config.txt.

JDasHO, 2021-10-20:
This worked wonders!
Thank you so much, I really apreciate it :D

6by9, 2021-12-03:
Update your kernel - various fixes went in for the touch controller in the last few weeks.

JDasHO, 2022-02-01:
Hi,
thanks for the hint.
I finally got to thoroughly test my setup with the kernel update (5.10.63-v7l+, from 1st Dec; loaded it as usually with apt-get update and upgrade). 
Unfortunately it doesn't solve the problems.

To be sure I tested the configuration of the fkms driver together with camera and dsi-touchscreen as well as the more promising configuration with the kms driver as you pointed out (last Oct., 19th). However for the latter I still find the same issue as described in my last comment that even though I can move the cursor with the touchscreen, I can't trigger click-signals. The cursor also for example keeps on holding this selection rectangle on the desktop as if the 'click' event only registered the pressed-event but not the released-event.

6by9, 2022-02-02:
None of that is the fault of raspiraw.

5.10.63 is not the latest kernel - apt was updated last week.

closed 2022-02-02 by 6by9