raspberrypi/raspiraw #62: RaspiRaw: No sensor found. Using a picamera v2 (imx219).

joram56, 2021-12-02:
On trying to follow [this](https://blog.robertelder.org/recording-660-fps-on-raspberry-pi-camera/), I am getting sensor not found after running `raspiraw` as below:

```
./raspiraw -md 7 -t 1000 -ts /dev/shm/tstamps.csv -hd0 /dev/shm/hd0.32k -h 64 -w 640 --vinc 1F --fps 660 -sr 1 -o /dev/shm/out.%06d.raw
Using i2C device /dev/i2c-0
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Probing sensor imx219 on addr 10
RaspiRaw: Probing sensor adv7282 on addr 21
RaspiRaw: No sensor found. Aborting
```

In `/boot/config.txt`, I have tried `dtparam=i2c_vc=on` and dtparam=i2c_arm=on (and restarted and ran `raspistill` before `camera_i2c`) with no success.

What am I missing?

6by9, 2021-12-02:
raspistill will reset all GPIO states and power the camera down. If you use it you will need to run ```./camera_i2c``` again.

> ```Using i2C device /dev/i2c-0```

If you read the output from ```camera_i2c``` it will have stated
```
Use i2c-10 for the sensor (-y 10)
````
You haven't included the ```-y 10``` options in your command line.

joram56, 2021-12-02:
I've tried with -y 10, get the same error.

[image]

6by9, 2021-12-02:
- Reboot your system
- Run ```./camera_i2c```
- Run ```sudo i2cdetect -y 10``` (if not installed, ```sudo apt install i2c-tools```). Report the addresses that are detected.

If an address reports as ```UU```, then you're running Bullseye where the kernel drivers are in use instead of the firmware. Remove the line ```camera_auto_detect=1``` from /boot/config.txt.

joram56, 2021-12-02:
Got this.

[image]

joram56, 2021-12-02:
After rebooting and running `i2cdetect -y 10`, everything started working. Thanks!

closed 2021-12-03 by 6by9