raspberrypi/raspiraw #67: "interface" folder not included in the Repository

hamedmh, 2022-03-18:
Hi, I'm trying to install raspiraw, using "make" but getting "dir/file not found" errors.
Thanks!

6by9, 2022-03-18:
It isn't part of this repo, it's in https://github.com/raspberrypi/userland/.
That would normally be installed as part of libraspberrypi0. Under Buster and before it would be in /opt/vc/, but Bullseye moves it to /usr/include which should be on the default search path anyway.

hamedmh, 2022-03-18:
Thank you @6by6 for the prompt reply.
I found the interface folder in /usr/include/, but it doesn't contain the mmal and vcsm dubfolders.
the interface folder of userland contains only mmal, not vcsm.
Is there a way to install the whole package? or complete the missing part?
Thanks!

6by9, 2022-03-18:
Ensure that you've updated your system. They were dropped when Bullseye was first released, but were added back in again.

Otherwise
```
git clone https://github.com/raspberrypi/userland
sudo apt install cmake
cd userland
./buildme
```
should install it all (although it goes into /opt/vc again).

hamedmh, 2022-03-18:
Thanks!
I have installed the system last week (arm64).
I tried installing userland again and checked all folders again and didn't see any change.

hamedmh, 2022-03-18:
Sorry, I didn't notice that the installation of userland was not performed correctly.
I saw errors related to missing vcsm, when building vcsm and user-vcsm.c.o

hamedmh, 2022-03-18:
Hi again,
userland is now successfully installed with your help. Thanks!
And now I need help with installing raspiraw.

I got the following result:

pi@pi:~ $ ls  ~/userland/interface/
**khronos  mmal  peer  vchi  vchiq_arm  vcos  vctypes  vmcs_host**

pi@pi:~ $ ls  /usr/include/interface/
**debug_sym  peer  vchi  vchiq_arm  vcos  vctypes  vmcs_host**

pi@pi:~ $ ls  /opt/vc/include/
**bcm_host.h  EGL  GLES  GLES2  IL  interface  KHR  vcinclude  VG  WF**

And when I try to make raspiraw I get the following error:

**pi@pi:~/raspiraw $  make** 
cc -I/opt/vc/include -pipe -W -Wall -Wextra -g -O0 -MD -c -o raspiraw.o raspiraw.c
In file included from raspiraw.c:50:
interface/mmal/util/mmal_util.h:31:10: fatal error: interface/mmal/mmal.h: No such file or directory
   31 | #include "interface/mmal/mmal.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:10: raspiraw.o] Error 1

6by9, 2022-03-19:
Brain not working - https://github.com/raspberrypi/userland/issues/688
MMAL is not supported with a 64bit userspace. raspiraw is therefore 32bit only.

closed 2022-03-19 by 6by9