raspberrypi/raspiraw #69: Errors when using tools, does not output files.

ruzko, 2022-04-20:
**Expected behaviour**
All commands execute without errors, and outputs files according to parameters.

**Actual behaviour**
Commands give errors and fail to output anything.

```
./640x64 1200 -ts tstamps.csv -o /dev/shm/out.%04d.raw
removing /dev/shm/out.*.raw
capturing frames for 1200ms with 659fps requested
ls: cannot access '/dev/shm/out.*.raw': No such file or directory
./640x64: line 14: 1000000 / : syntax error: operand expected (error token is "/ ")
frame delta time[us] distribution
after skip frame indices (middle column)
expr: division by zero
% frame skips
```

**Environment**
Raspberry Pi 4B
Raspbian GNU/Linux 11, kernel 5.15.32
Camera sensor: IMX219
Legacy camera support is turned on. 
[Userland libraries](https://github.com/raspberrypi/userland) installed

Output of inxi -Faz:

```
System:    Kernel: 5.15.32-v7l+ armv7l bits: 32 compiler: gcc v: 10.2.1
           parameters: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0
           snd_bcm2835.enable_hdmi=1 smsc95xx.macaddr=DC:A6:32:E0:15:7E vc_mem.mem_base=0x3ec00000
           vc_mem.mem_size=0x40000000 console=ttyS0,115200 console=tty1 root=PARTUUID=7b40f466-02
           rootfstype=ext4 fsck.repair=yes rootwait
           Console: tty 0 Distro: Raspbian GNU/Linux 11 (bullseye)
Machine:   Type: ARM Device System: Raspberry Pi 4 Model B Rev 1.4 details: BCM2711 rev: d03114
           serial: <filter>
CPU:       Info: Quad Core model: ARMv7 v7l variant: cortex-a72 bits: 32 type: MCP arch: v7l
           family: 7 model-id: N/A stepping: 3
           features: Use -f option to see features bogomips: 1296
           Speed: 1800 MHz min/max: 600/1800 MHz Core speeds (MHz): 1: 1800 2: 1800 3: 1800
           4: 1800
           Vulnerabilities: Type: itlb_multihit status: Not affected
           Type: l1tf status: Not affected
           Type: mds status: Not affected
           Type: meltdown status: Not affected
           Type: spec_store_bypass status: Not affected
           Type: spectre_v1 mitigation: __user pointer sanitization
           Type: spectre_v2 status: Vulnerable
           Type: srbds status: Not affected
           Type: tsx_async_abort status: Not affected
Graphics:  Device-1: bcm2711-vc5 driver: vc4_drm v: N/A bus ID: N/A chip ID: brcm:gpu
           class ID: gpu
           Device-2: bcm2711-hdmi0 driver: N/A bus ID: N/A chip ID: brcm:soc class ID: hdmi
           Device-3: bcm2711-hdmi1 driver: N/A bus ID: N/A chip ID: brcm:soc class ID: hdmi
           Display: server: No display server data found. Headless machine? tty: 102x59
           Message: Advanced graphics data unavailable in console. Try -G --display
Audio:     Device-1: bcm2835-audio driver: bcm2835_audio bus ID: N/A chip ID: brcm:bcm2835_audio
           class ID: bcm2835_audio
           Device-2: bcm2711-hdmi0 driver: N/A bus ID: N/A chip ID: brcm:soc class ID: hdmi
           Device-3: bcm2711-hdmi1 driver: N/A bus ID: N/A chip ID: brcm:soc class ID: hdmi
           Sound Server: ALSA v: k5.15.32-v7l+
Network:   Message: No ARM data found for this feature.
           IF-ID-1: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter>
           IF-ID-2: wlan0 state: up mac: <filter>
Drives:    Local Storage: total: 14.46 GiB used: 1.86 GiB (12.9%)
           SMART Message: Required tool smartctl not installed. Check --recommends
           ID-1: /dev/mmcblk0 maj-min: 179:0 model: SD16G size: 14.46 GiB block size:
           physical: 512 B logical: 512 B rotation: SSD serial: <filter> scheme: MBR
Partition: ID-1: / raw size: 14.2 GiB size: 13.95 GiB (98.21%) used: 1.81 GiB (13.0%) fs: ext4
           dev: /dev/mmcblk0p2 maj-min: 179:2
           ID-2: /boot raw size: 256 MiB size: 252 MiB (98.46%) used: 49.2 MiB (19.5%) fs: vfat
           dev: /dev/mmcblk0p1 maj-min: 179:1
Swap:      Kernel: swappiness: 60 (default) cache pressure: 100 (default)
           ID-1: swap-1 type: file size: 100 MiB used: 0 KiB (0.0%) priority: -2 file: /var/swap
Sensors:   System Temperatures: cpu: 46.7 C mobo: N/A
           Fan Speeds (RPM): N/A
Info:      Processes: 128 Uptime: 31m Memory: 7.79 GiB used: 577.8 MiB (7.2%) gpu: 256 MiB
           Init: systemd v: 247 target: multi-user.target Compilers: gcc: 10.2.1 alt: 10 Packages:
           apt: 601 lib: 330 Shell: Bash v: 5.1.4 running in: tty 0 (SSH) inxi: 3.3.01
```

6by9, 2022-04-20:
Reporting ```supported=0``` on a Pi4 implies you've disabled the camera stack totally.
What does ```vcgencmd version``` report?

ruzko, 2022-04-20:
Ran a fresh image of raspbian buster with kernel 5.10. 
Verified that raspistill works and raspiraw produces the same errors, so I'm reopening the issue.

On this new image, `vcgencmd version` reports
```
Dec  1 2021 15:02:46
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start_x)
```
On a side note, using this fresh image, `vcgencmd` reports `supported=1 detected=1` , so it's likely as you say that I misconfigured the previous OS image.

closed 2022-04-22 by ruzko