raspberrypi/raspiraw #70: Can't increase framerate above 380 fps [Rpi4+IMX219]

ruzko, 2022-04-27:
Why can't raspiraw exceed 380 fps?

running `raspiraw -md 7 -t 100 -y 10  -ts /dev/shm/tstamps.csv -hd0 /dev/shm/hd0.32k -sr 1 -o /dev/shm/out.%04d.raw --fps 380` makes 38 `out.*.raw` files, 1 `hd0.32k` file and 1 `tstamps.csv` file.
All 38 out.*.raw files are valid.

```
cat tstamps.csv
,1,2922479898
2610,2,2922482508
2609,3,2922485117
2610,4,2922487727
2609,5,2922490336
2610,6,2922492946
2609,7,2922495555
2610,8,2922498165
2609,9,2922500774
2610,10,2922503384
2609,11,2922505993
2610,12,2922508603
2609,13,2922511212
2610,14,2922513822
2609,15,2922516431
2610,16,2922519041
2609,17,2922521650
2610,18,2922524260
2609,19,2922526869
2610,20,2922529479
2609,21,2922532088
2610,22,2922534698
2609,23,2922537307
2610,24,2922539917
2609,25,2922542526
2610,26,2922545136
2609,27,2922547745
2610,28,2922550355
2609,29,2922552964
2610,30,2922555574
2609,31,2922558183
2610,32,2922560793
2609,33,2922563402
2610,34,2922566012
2609,35,2922568621
2610,36,2922571231
2609,37,2922573840
9223372033932201968,38,-9223372036854775808
-9223372036854775808,39,0
0,40,0
```

-----------------------------------------------
running `raspiraw -md 7 -t 100 -y 10  -ts /dev/shm/tstamps.csv -hd0 /dev/shm/hd0.32k -sr 1 -o /dev/shm/out.%04d.raw --fps 400` makes 7 `out.*.raw` files,  1 `hd0.32k` file and 1 `tstamps.csv` file.
Only `out.0001.raw` is valid, the remaining 6 `out.*.raw` files are empty.

```
cat tstamps.csv
,1,2699571955
9223372034155203853,2,-9223372036854775808
0,3,-9223372036854775808
-9223372036854775808,4,0
0,5,0
0,6,0
0,7,0
```
-----------------------------
`/dev/shm` has enough capacity:
```
df -h /dev/shm
Filesystem      Size  Used Avail Use% Mounted on
tmpfs           7.0G   12M  7.0G   1% /dev/shm
```

```
inxi -S
System:
  Host: raspberrypi Kernel: 5.10.103-v7l+ armv7l bits: 32 Console: tty 0
  Distro: Raspbian GNU/Linux 10 (buster)
```
`board revsion: d03114 `

6by9, 2022-04-27:
All those frame deltas are ~2609usecs, therefore there is no issue of frames being dropped. It is saving data as it is being received.

Sensor configuration is done over I2C. Please note https://github.com/raspberrypi/raspiraw/blob/master/imx219_modes.h#L29
```
// These values are copied from
// https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=162722&p=1052339
// Raspberry Pi can not discuss these settings as we have information from
// Sony under NDA.

// REQUESTS FOR SUPPORT ABOUT THESE REGISTER VALUES WILL
// BE IGNORED BY PI TOWERS.
```

There is a datasheet for IMX219 kicking around on the internet, so the registers can be analysed.
I suspect that the fps calcs haven't clipped the vts values correctly, although there was also a fix in the imx219 kernel driver as https://github.com/raspberrypi/linux/commit/064d872084395d74bba329cbe1cca1e34354cd3a which corrected the min vblank value from 4 to 32, which would be 512 at https://github.com/raspberrypi/raspiraw/blob/master/imx219_modes.h#L696

ruzko, 2022-04-27:
Thanks @6by9, I'll try to analyse the registers.
Changed L696 from 484 to 512 and ran `make clean && make` but observed no change in behaviour.