raspberrypi/raspiraw #78: Cant get it to work with rp3 model b v1.2

Hilldamn, 2023-07-14:
Hi I tried to run the ./camera_i2c command and I get this back

setting GPIO for board revision: a02082
Rasberry Pi3B / Pi3B+ / 3A
Set state of 133 to 1
Use i2c-10 for the sensor (-y 10)

When I then try running from inside the directory ~/raspiraw/raspiraw -md 1 -y 10

Using I2C device /dev/i2c-10
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Found sensor ov5647 at address 36
RaspiRaw: Encoding 41414770
No AWB
mmal: mmal_vc_component_create: failed to create component 'vc.ril.rawcam' (5:ENOENT)
mmal: mmal_component_create_core: could not create component 'vc.ril.rawcam' (5)
mmal: Failed to create rawcam

I am using Rasbian GNU Linux 11 bullseye.

6by9, 2023-07-14:
raspiraw is dependent on the legacy camera stack, whilst Bullseye has switched to libcamera.
I really ought to find time to rewrite raspiraw to use the dummy_sensor driver and V4L2 to work in the new ways.

You'll need to add "start_x=1" and "gpu_mem=128" to /boot/config.txt, and remove "camera_auto_detect=1" to get the legacy stack working.

Hilldamn, 2023-07-14:
Hi thanks for the quick reply!

after editing the config file and running ~/raspiraw/raspiraw -md 1 -y 10 again I get this message

Using I2C device /dev/i2c-10
RaspiRaw: Probing sensor ov5647 on addr 36
RaspiRaw: Found sensor ov5647 at address 36
RaspiRaw: Encoding 41414770
No AWB
mmal: Set pack to 0, unpack to 0
mmal: Timing 6/2, 2/6/0, 0/0
mmal: Create pool of 6 buffers of size 2611200
mmal: Create pool of 6 buffers of size 2611200
mmal: Now streaming...
mmal: mmal_port_pool_destroy: port 0x182160, pool 0x188d80 destroyed while port enabled

should I be concerned?

6by9, 2023-07-14:
> should I be concerned?

Which message are you concerned about?

```mmal: mmal_port_pool_destroy: port 0x182160, pool 0x188d80 destroyed while port enabled``` is the only error, and that's just that the cleanup code isn't correct. Seeing as the application is quitting, that's not an issue.

Hilldamn, 2023-07-14:
Ah I see. Thanks for the help I'll try recording some footage now and see if it works

closed 2023-07-14 by Hilldamn