raspberrypi/raspiraw #80: 240FPS on IMX477

huyleincepit, 2024-06-02:
Hello,

I am currently developing an application that requires a high-speed camera. I am using IMX477 and Jetson Nano, but the Jetpack Driver only supports 60fps. I am attempting to achieve 1348x750 resolution at 240fps. I plan to update the Jetpack Driver, but I am unable to locate the datasheet for IMX477. Could you kindly provide me with the full datasheet for IMX477, including detailed registers?

Thank you,
Huy Le

6by9, 2024-06-02:
Whilst the comment didn't get copied from imx219_modes.h into imx477_modes.h, I'm afraid it is also valid for imx477
https://github.com/raspberrypi/raspiraw/blob/master/imx219_modes.h#L29-L35

The datasheet is under NDA, so we can not distribute it.

closed 2024-06-02 by 6by9

6by9, 2024-06-07:
FWIW unrelated to this I've just been tweaking IMX477 settings for Pi5. 

https://github.com/raspberrypi/linux/pull/6208 results in a 1332x990 raw10 mode that should be able to achieve 254fps. I provide no guarantees of compatibility with the Jetson though.